Ultrawipf/OpenFFBoard #1: what would you like help with?

baitnfatty, 2020-04-14:
Yannik, 
I have been doing research for this project the last few weeks and was just about to start the same adventure with servo motors as it made me upset to see all of these people starting the projects seeing its involvement and then selling the boards for 300-400$ when they all used open source software. so i greatly appreciate your dedication to your work and would like to help you in any way possible.

I have not programmed in python for a while as most my recent stuff has been in C# but it would not be hard to pick up again. it appears you already started working on the GUI and everything so what left do you have to do?

It was hard to get a grasp of what you were saying you were having issues with in the video. is the FFB working 100% ?  were you able to figure out the interface and how the drivers work or do you still need help with that?  

I look forward to hearing back from you and would like to exchange emails 
I can be reached at mgriffizl39 ( AT) gmail(DOT)com

Ultrawipf, 2020-04-14:
Yes the FFB works pretty well actually and its usable. At the moment i am mostly trying to get a servo for further testing before making more changes in the hardware.
The GUI is written in python and at the moment very basic and could need a bit more structure and error handling.
I will contact you by mail for more details.

baitnfatty, 2020-04-15:
Thanks for the response. I look forward to your email. Today i just kinda looked over the files to get an idea of how you have things structured. looking forward to talking more in depth.

Scorb1, 2020-04-24:
This looks interesting to me as well, I ran across this recently but I'm interested in helping.

kai0076, 2020-06-07:
Hello Yannick,
I found your project here and like to support you with this project. How can we get in touch via email, I cannot find your email adress here in GitHub. Could you let me know your email or how to find it here?
Looking forward.
Best regards
Kai

tinyboxxx, 2020-06-07:
hi, welcome. Join us at discord:  https://discord.gg/jhCQQvc

On 06/07/2020 19:31, kai0076 wrote:

Hello Yannick,
I found your project here and like to support you with this project. How can we get in touch via email, I cannot find your email adress here in GitHub. Could you let me know your email or how to find it here?
Looking forward.
Best regards
Kai

closed 2021-09-05 by Ultrawipf