Ultrawipf/OpenFFBoard #14: info servo motor

pipes80-collab, 2021-02-06:
good evening everyone and nice project.
I state that I am not an expert but I would like to try my hand at a direct drive station to replace my g29 and try to spend little.
These compatible servo motors look great.
I start with the questions :-)
1. Does this servo motor need to be purchased with its servo driver? see link: https://it.aliexpress.com/i/32720870362.html
2. Can other servo motors not in the list be compatible and not create too many problems with in-game compatibility?
Thank you very much and good job

Ultrawipf, 2021-02-06:
Power consumption depends on motor and how intense the force should be. On average it does not pull a lot but for example if you have a lower torque motor you may pull 300W+ for a short time while a 130st 15015 or 10010 would do the same torque with only 50W. Overall it depends on how you drive it and how much force you expect and obviously on the motor. 
You can use other motors as well as long as they have an ABN or SinCos encoder.

You can probably not use the driver. The TMC driver made for this project is the preferred choice but if you have a different driver which has a PWM controlled torque mode it would be easy to use as well but maybe not as smooth. The linked driver i think has quite a low resolution and needs an analog voltage in torque mode so i would not use that. You may be able to make it work with a circuit generating the correct analog voltage from a PWM value but it will probably be quite noisy or slow.

I would recommend you to hop in our discord to keep up to date and see what others are making.

pipes80-collab, 2021-02-06:
so better to use a servo motor like this?  
https://www.amazon.it/dp/B086PDFFP5/ref=cm_sw_r_cp_apa_i_DVKMS3S5KFWH5H927CN7
and then connect it to the TMC driver?

Ultrawipf, 2021-02-06:
Thats a stepper and not a servo motor. Will still work as the tmc driver also has stepper support but a 3 phase servo is much smoother.