Ultrawipf/OpenFFBoard #16: Implement missing FFB effects

Ultrawipf, 2021-03-02:
Most effects are not used by most games but all effects reported as supported should be implemented in some way.

Supported are currently:
Constant force
Sine
Square
Spring
Damper (friction currently same formula)
Deadbands for spring and damper

Missing: 
Correct friction calculation
Inertia effect
Triangle
Ramp settings
Optional custom force (or remove from descriptor as its not really standard anymore)

geek-slt, 2021-03-30:
I did not find the relevant calculation formula in the USB PID document. Does the force feedback calculation formula need to be implemented by myself?

Ultrawipf, 2021-03-30:
> 
> 
> I did not find the relevant calculation formula in the USB PID document. Does the force feedback calculation formula need to be implemented by myself?

There are some examples how to implement the conditional effects. But its already being worked on in the active development branch.

geek-slt, 2021-03-30:
How long will it take to perfect these force feedbacks? I will upload a document later, which contains the types of force feedback output by different games.

Ultrawipf, 2021-03-30:
Well. Perfect depends on the game and hardware. But the goal is to make it as tunable as possible and provide a good feel with most setup.
Overall i think it is pretty good when tuned properly but it needs a lot of testing and comparing to further improve it.

Having a list of games and which effects they use would be great. 
Maybe also for what scenario a different effect is used if possible (like spring only in menu, or sine for extra rumble) and at which rate updates are sent if known.

geek-slt, 2021-03-30:
test result：
[image]

geek-slt, 2021-03-30:
From the results, Friciton needs to be supported. Traingle SawtoothUp SawtoothDown is unnecessary.

jonlidgard, 2021-03-30:
All effects apart from custom are already implemented, although there is stills some tweaking to do with a few of them (friction, inertia - rarely used). If you use the latest dev version (rtos_multiaxis) you can run a game & then, using the Configurator app, issue the 'effects' command & it will tell you which effects the game used. Essentially any game that respects the official FFB spec should work, some work better than others but that is down to how the game is coded & not something we can change.

jonlidgard, 2021-03-30:
Friction is supported but we're still experimenting with different gains & filteringt to improve the feel.

geek-slt, 2021-03-30:
Ok, I will try the rtos_multiaxis branch. I am currently modifying the code in the Configurator and the main branch to get the effect of the game.

jonlidgard, 2021-03-30:
One thing that isn't implemented yet is assigning an effect to a button e.g. pressing 'fire' button gives feedback which might get used in some games like DCS

geek-slt, 2021-03-30:
Does the program need to adjust the gain and filtering for different motors to improve the feel?

jonlidgard, 2021-03-30:
> Ok, I will try the rtos_multiaxis branch. I am currently modifying the code in the Configurator and the main branch to get the effect of the game.

No  need, that functionality is already implented in the dev branch. (You'll need the rtos_multiaxis branch of Configurator too)

jonlidgard, 2021-03-30:
Obviously different motors will require different tuning according to user prefs, etc.

geek-slt, 2021-03-30:
Okay, got it

geek-slt, 2021-03-30:
> 是的，我一直在为XPlane提供多轴支持和测试方面的帮助-尽管仍处于初期阶段，但仍需要解决一些问题，以解决是否需要连接多个驱动器板的问题

discard --->arilink#4555

geek-slt, 2021-04-09:
@jonlidgard  Hello, I am trying to use rt_multiaxis configurator and running run.bat prompts No moudle named ‘intelhex’. I tried to install intelhex on windows but failed. Could you please ask for your help?

PR #24 (2021-05-22): V1.3 Release - FreeRTOS + Tinyusb + Multiaxis support
Major changes:

- System ported to support FreeRTOS (with C++ helpers)
- Replacing old modified ST HAL based USB system with TinyUSB (Rtos safe, easier management)
- Improved TMC4671 init sequence (Closed loop current mode, safer)
- Basic error handling
- Multiple FFB axes possible
- Reworked encoder management
- Implemented all HID FFB effects ( Closes #16 )
- Classchooser can check and report to gui if a class can't be created due to peripheral conflicts for example

While bringing a lot of improvements remember that this is a large updat [... cut]
closed 2021-05-22 by Ultrawipf