Ultrawipf/OpenFFBoard #20: Is it compatible with European Truck Simulator 2?

TDA-2030, 2021-03-17:
I have compiled the code with CubeIDE and download firmware to STM32F407, then connect the board to pc with usb and hear a tone from pc. But, I can't find new control device in setting of European Truck Simulator 2. By the way, I can see a new serial device in windows device manager panel after connect board.

So, I  want to know that this project is compatible with European Truck Simulator 2? Is there a list of supported games?

tinyboxxx, 2021-03-17:
Please try the GUI first, there is some configuration needed.

BTW, if you use Chinese, you can have a look in our (unofficial) QQ chat group:1108176152

jonlidgard, 2021-04-14:
Yes, I did a quick test - EuroTruck Simulator 2 uses the following effects: Square,Sine,Spring,Friction , all of which are modelled by openFFBoard.

closed 2021-04-14 by jonlidgard