Ultrawipf/OpenFFBoard #51: First Try

vtrx-az, 2022-03-06:
I compiled using Cube.
I have the STM Discovery card (STM32F407), is the firmware compatible?
I loaded the hexadecimal on the board, Windows installed the drivers and recognized the board, but in the Joystick manager it does not appear.
Something is missing?

felixmxr, 2022-03-06:
Pin PA9 needs to be pulled high (3.3v) to activate USB functionality.
Does the debugger connection work on your board?

vtrx-az, 2022-03-06:
STM Discovery card PA9 is at a high level.
As I mentioned, Windows installed and recognized it as a FFB device, but in the device manager the Joystick does not appear.
Is there any further configuration needed?

vtrx-az, 2022-03-06:
Solved.
it is necessary to use the configurator.

closed 2022-03-25 by Ultrawipf