Ultrawipf/OpenFFBoard #87: Operating system support

HiranChaudhuri, 2023-01-14:
What operating systems does the firmware work with? What is it being tested on?

Ultrawipf, 2023-01-15:
Mainly developed and tested for Windows but it is also being tested successfully on Linux (behaves differently but most effects work under wine) and some users use it successfully on Mac as well.
Any possibly USB/Effect compatibility related change is always tested on both Windows and Linux.

HiranChaudhuri, 2023-01-15:
Thank you for coming back to me on this. This information is for sure worth to be placed in the documentation.
Since I am running Linux: What differences are to be expected?

Ultrawipf, 2023-01-16:
On linux the device gets immediately set into active FFB mode. so no automatic idle spring effect outside games and some wine versions did not support all effect modes and only very few native games support full FFB at all. In general FFB compatibility on Linux is very sensitive to the HID descriptor as well but at the moment on my test setup it seems to work fine.
Running the iracing wheelcheck or fedit under wine allowed me to control most effects correctly and some native ffb test tools worked as well.

HiranChaudhuri, 2023-01-16:
The fact that only few games on Linux support ffb can hardly be solved within the scope of this project. But the joystick should be somewhat compatible with the driver such that it behaves similar as the other ffb joysticks. I own the Microsoft Sidewinder Force Feedback 2, and that one stays calm (like a standard joystick) until force feedback commands are issued.

So far I ran them using `fftest` and `ffcfstress`. But then it is likely that my joystick is talking the Microsoft Digital Overdrive protocol (see https://www.kernel.org/doc/html/latest/input/joydev/joystick.html#microsoft-sidewinder-joysticks) which may not be the case for OpenFFB. Which protocol/kernel driver is actually used? Is there a way to debug such communication?

Ultrawipf, 2023-01-18:
It is a standard USB Physical interface device (as in windows direct input compatible). So it is using hid-pidff which is very sensitive to the descriptor layout so a lot of devices may not work if optional fields are missing. I spent quite some time debugging that and getting it to work on both windows and linux so at the moment it should be fine.

One of the threads about the differences is here: https://github.com/berarma/ffbtools/issues/26 but there seem to be some talks about improving the support.

HiranChaudhuri, 2023-01-18:
Oh, it seems the devil is in the details. Maybe it is not that easy to achieve homgeneous compatibility.
I will consider this when I really go for a DIY FF device.

Thank you for sharing that much insight. :-)

closed 2023-01-18 by HiranChaudhuri