Ultrawipf/OpenFFBoard #114: Compilation error due to missing DRV_BRAKE_PIN even when TMC4671 driver Is commented uut

OrhanYigitDurmaz, 2024-09-28:
I am working on creating a new target for the f405rgt6, but I don’t have enough pins. As a result, I am cutting off unnecessary components, such as PWM, UART, SPI, and the TMC4671 driver. However, when I try to compile, I encounter the following error:

`error: 'DRV_BRAKE_PIN' is not defined`

This happens because, even though the TMC4671 driver is commented out, the build process is still looking for the DRV_BRAKE_PIN, which I haven’t defined. The VoltageSense.cpp file should only be compiled when the TMC4671 driver is defined, but it seems that it is being compiled regardless of this condition.

# Steps to Reproduce:

Comment out the TMC4671 driver in the target configuration.
Attempt to compile the project.
Observe the compilation error regarding the missing DRV_BRAKE_PIN.

# Expected Behavior:

The build process should not look for DRV_BRAKE_PIN when the TMC4671 driver is disabled. The VoltageSense.cpp should only be compiled when TMC4671 is defined.
# Possible Solution:

Add conditional compilation to VoltageSense.cpp so that it only compiles when the TMC4671 driver is defined. Alternatively, ensure that any dependencies on DRV_BRAKE_PIN are wrapped in preprocessor conditionals based on the TMC4671 definition.

Ultrawipf, 2024-11-15:
As mentioned in the PR: the brake resistor pin is still useful without the TMC4671. A better solution would be to make the voltage sensing stuff depend on the pin definition but as system commands and global ADC configurations depend on that it is much more than a simple change so i will keep it in mind when rewriting the ADC management but won't merge it like it is yet.

closed 2024-11-15 by Ultrawipf

OrhanYigitDurmaz, 2024-11-15:
> .
Thats the better fix of course. This was just a temporary fix to use F405. I was trying to create a new target but that doesnt work because f405 doesnt have all the pins that f407 has. What ill do now is to just copy the f407 target, and change the pins according to f405 pinout. This way FFBoard just writes to missing registers. This should work better as a temporary solution.