Ultrawipf/OpenFFBoard #120: Crashed while launch Forza Horizon 5

hirex, 2024-12-18:
I followed the game setup instructions provided in the wiki, but when I launched Horizon 5, the game crashed.

Game setup:
[image]

Upon checking the Windows Event Log, I found the following error details:
Description:
Error application name: ForzaHorizon5.exe Version: 1.671.65.0, timestamp: 0x00000000
Module name: pid.dll, version: 10.0.19041.1, timestamp: 0xf385cf53
Exception code: 0xc0000005
Error offset: 0x000000000000266e
Error process ID: 0x118c
Error application start time: 0x01db50f9b8a7276a
Error application path: D:\Program Files (x86)\Steam\steamapps\common\ForzaHorizon5\ forzahorizon5.exe
Error module path: C:\Windows\System32\pid.dll
Report ID: cf96bbe6-eb88-4d86-a2fe-3e1fe844f326

Has anyone else encountered this issue? Can you provide guidance on how to resolve it?

Additionally, I’m unsure how the ID in the Horizon 5 game profile, such as:
id="a5dcbf10-6530-11d2-901f-00c04fb951ed",
is obtained. Could you explain the process for retrieving this ID?

Ultrawipf, 2024-12-19:
Forza crashes at every opportunity with many devices but the FFBoard should not cause that. It may require reconnection after starting the game but it should work. Could it be a different device causing it? 
Does it always happen only with the FFBoard connected? Try starting it only with one device to isolate which one crashes it.

hirex, 2025-01-14:
Yes, always crashes in the startup game. I looked up some information and it has been fixed, it was the game itself, thanks.

closed 2025-01-25 by Ultrawipf