Ultrawipf/OpenFFBoard #122: add supoted games

nhy6iii, 2025-01-09:
openFFBoard works with "XPforce" without any modification.

XPForce is a plugin for X-Plane from Laminar Research, Microsoft Flight Simulator 2020 and 2024, Microsoft/Dovetail FSX or FSX-SE, Flight Sim World, and Lockheed Martin Prepar3D v4. XPForce adds support for force-feedback joysticks, yokes and pedals on Windows!

Ultrawipf, 2025-02-25:
Currently testing different flight sims with 2 axis joystick setups and there seem to be a few issues with the direction handling that need to be improved.

XPForce for some reason does not work for me. X-Plane crashes when XPForce is installed.

Ultrawipf, 2025-03-21:
Update: Direction handling has been corrected. It should now work the same way as the MSFFB stick.

xpforce does not work in the steam versions of xplane for some reason but it fine with the standalone xplane12 version.

closed 2025-03-21 by Ultrawipf