Ultrawipf/OpenFFBoard #127: Add commands to get query the ODrive's FW and use that determine what error messages to display in the GUI

datapaganism, 2025-02-18:
Myself and others who use an Odrive use a 3.6 clone are stuck on 0.5.6 firmware of the Odrive

Troubleshooting errors can be quite difficult currently since the GUI is mapped to post 0.5.6 error codes, leading to confusion

I can make changes to odrive_ui.py to so it uses 0.5.6 mappings but that hurts users who use later revision Odrives

I wondering if we could add support to the MCU to query the FW version of the Odrive over CAN and report that to the GUI on init
so we can switch dynamically between different error code mappings

datapaganism, 2025-02-18:
Or a toggle just on the GUI and skip the MCU all together

Ultrawipf, 2025-02-19:
It would need to be detected in a different way. There is no command to check the version in the old CAN protocol. In the new one there is a bit more available so maybe a lack of a reply could be used to find out the version...

Otherwise for troubleshooting the odrivetool should be used. That makes it easier anyways. 
The display in the GUI is there just because the error code is sent but initially it just showed the int value of it and could be removed completely if it is more confusing than helpful.

datapaganism, 2025-02-20:
Okay I think it's best if we just render the error code (and in hex as well)

Rather than printing the error string which is the misleading bit

Its a bit quicker for me to open up the Odrive docs than find a microusb and open up odrivetool and remember how to dump the errors